pyvisa/pyvisa #824: [COM] Communication issue with ROHDE&SCHWARZ EMI test receiver using LAN and DHCP Networking setting

Jonjinadsf, 2024-05-21:
Instrument details
 #------------------
* Model: ESR-3
* Communication: TCPIP
* Link to the documentation (if available):
    https://scdn.rohde-schwarz.com/ur/pws/dl_downloads/dl_common_library/dl_manuals/gb_1/e/esr_1/ESR_UserManual_en_13.pdf

Output of `pyvisa-info`
-----------------------

Machine Details:
   Platform ID:    Windows-11-10.0.22631-SP0
   Processor:      Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\Users\v-zhixianjin\OneDrive - Microsoft\Desktop\EMC Test Automation\Tool\.venv\Scripts\python.exe
   Version:        3.12.3
   Compiler:       MSC v.1938 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Apr  9 2024 14:05:25 (#tags/v3.12.3:f6650f9)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 25165824
         Spec. Version: 7340544
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 25165824
         Spec. Version: 7340544

Jonjinadsf, 2024-05-21:
Hello, the template is filled as above. Pyvisa cannot detect the device I connect sometimes. What I find is that I need to make sure the NI-MAX client can detect the device so pyvisa can detect it. NI-Max can use auto detect to find the device and recover communication and I wonder is there a way for pyvisa to do samething or not. Thank you.

[image]

MatthieuDartiailh, 2024-05-22:
PyVISA simply pipes the information it gets from NI in your case. So if you need to manually add the device to NI there is little we can do on the PyVISA side. However even if you cannot list the device you should be able to connect to it since it appears to be present since you can add it.

DavidLutton, 2024-05-22:
I use PyVISA-py to discover most of our networked instruments.

Is your instrument listed when you run, this?:
``` python
from pyvisa_py.tcpip import TCPIPInstrVxi11 as TCPVXI11

print(TCPVXI11.list_resources())
```

MatthieuDartiailh, 2024-05-22:
Out of curiosity why did you choose to do it that way rather than creating a `ResourceManager("@py")` and then use `list_resources`

Jonjinadsf, 2024-05-22:
Hello Matthieu, David code works for me, and I will also try to use the "@py" method you mentioned. my guess both method is getting the backend of NI VISA and grab information. My assumption in my first post is that: I found NI-Max has way to detect instrument that does not show up in the first place. and there might be a way pyvisa can do that too.  "rm.list_resource()" is not stable for me sepcially when I unplug the device and plug it back again, it will fail to detect the device. 
Thank you for your input and quick response!

MatthieuDartiailh, 2024-05-27:
Using PyVISA-py fully bypass NI. 

I am happy you found a working solution. I will close. Feel free to re-open if needed.

closed 2024-05-27 by MatthieuDartiailh

DavidLutton, 2024-06-02:
@MatthieuDartiailh sorry I didn't get back to this
I use a mix of GPIB, USB and LAN equipment for testing/calibrations
So resource discovery starts with:
``` python
import pyvisa
from pyvisa_py.tcpip import TCPIPInstrVxi11 as TCPVXI11  # TCPVXI11.list_resources()

ignores = ['GPIB0::22::INSTR','TCPIP::192.168.100.5::INSTR',]

rm = pyvisa.ResourceManager()
resources = list(rm.list_resources('(GPIB[012]::?*::INSTR)|(USB?*)')) + TCPVXI11.list_resources()
resources = [resource for resource in resources if resource not in ignores]
```
Followed by probing for discovery of the make & model, then by driver assignment in [LabToolkit](https://github.com/DavidLutton/LabToolkit/blob/main/src/labtoolkit/__init__.py)
Including some pre `*IDN?` hardware that give hints in its errors