pyvisa/pyvisa #826: [COM] Communication issue with HP 8720B using GPIB - Cannot read full frequency sweep

leonknight99-flann, 2024-05-23:
Instrument details
------------------
* Model: HP 8720B
* Communication: GPIB
* Link to the documentation (if available):
https://docs.alltest.net/manual/Alltest-HP-Agilent-Keysight-8720B-Quick%20reference%20guide.pdf
https://aolab.fer.hr/_download/repository/Manual_hp8720B_Programming.pdf

Output of `pyvisa-info`
-----------------------

Hi,

I am having trouble reading the full frequency sweep for my VNA. I can control the VNA with the basic commands such as setting the start and stop frequencies, number of points, channel settings etc but when I want to output the real and imag data points over my frequency range it falls over and only outputs the first one. 

```
rm = pyvisa.ResourceManager()
my_instrument = rm.open_resource('GPIB0::16::INSTR')
my_instrument.read_termination = '\n'
my_instrument.timeout = 60000
print(my_instrument.query('*IDN?'))

my_instrument.write("PRES;")  # Presets VNA

my_instrument.write("CHAN1; S11; LOGM;")  # Sets Channel 1 to S11 LogMag
my_instrument.write("CHAN2; S11; PHAS;")  # Sets Channel 2 to S11 Phase

my_instrument.write("DUACON")  # Dual Channel Display

my_instrument.write("STAR 10 GHZ;")  # Sets min frequency
my_instrument.write("STOP 15 GHZ;")  # Sets max frequency
my_instrument.write("POIN 11")  # Sets number of points

start_freq = my_instrument.query("STAR?")
stop_freq = my_instrument.query("STOP?")

print(start_freq)
print(stop_freq)
my_instrument.write("POIN 11; FORM4; OPC?; SING;")

dataArray = my_instrument.query_ascii_values("OUTPDATA;", container=np.array)

print(f'ASCII {dataArray} {-20*np.log10(np.sqrt(dataArray[0] ** 2 + dataArray[1] ** 2))}') 
```

Returns: 
ASCII [-0.8023682 -0.2625732] 1.4706922451729465

So returns a 1x2 array instead of what I want, a 11x2 array.

Thanks!

MatthieuDartiailh, 2024-05-27:
Can you try reading one byte at a time (using `read_bytes(1)`) and print it so that we know what the instrument returned ?

DavidLutton, 2024-06-02:
@leonknight99-flann considering your line `my_instrument.write("POIN 11; FORM4; OPC?; SING;")`
Does reordering to `my_instrument.write("FORM4; SING; OPC?")` make a difference
Drop POIN 11 as is also a few lines up

I think that ending with SING should start new measurement
Then the instrument only passes back the data it has ready at the time you query for OUTPDATA

if that works gets your data try out
``` python
inst.query_ascii_values('OUTPDATA;', container=np.float64).view(np.complex128)
# Then you can use
-20 * np.log10(np.sqrt(dataArray.real**2 + dataArray.imag**2))
# or you can use
magnitude = np.abs(dataArray)
-20 * np.log10(magnitude)
```

leonknight99-flann, 2024-06-03:
@MatthieuDartiailh I added a `read_bytes(1)` before and after the `query_ascii_values()` and have received the output:

HEWLETT PACKARD,8720B,0,1.01
   1.000000000000000E+10
   1.500000000000000E+10
b'1'
b' '
ASCII [0.04120827 0.0350914 ] 25.332023659333384

Not sure if that says anything in particular - I have now connected the input and output ports together which is why the parameter has changes.

`
start_freq = my_instrument.query("STAR?")
stop_freq = my_instrument.query("STOP?")

print(start_freq)
print(stop_freq)
my_instrument.write("FORM4; SING; OPC?;")
print(my_instrument.read_bytes(1))

dataArray = my_instrument.query_ascii_values("OUTPDATA;", container=np.array)
print(my_instrument.read_bytes(1))

print(f'ASCII {dataArray} {-20*np.log10(np.sqrt(dataArray[0] ** 2 + dataArray[1] ** 2))}') 
`

leonknight99-flann, 2024-06-03:
@DavidLutton Unfortunately it does not seem to make any difference which way round I have the `"SING;"` command and doesn't make any difference if I enter `"CONT;"` for a continuous sweep instead. However, I have found it works using the `"FORM2;"` output instead for some reason. Will keep trying different commands.

MatthieuDartiailh, 2024-06-03:
The fact that your second read_bytes does not timeout is suspicious. You could try to keep reading till it does not get a better idea of the buffer content.

As a side node you are likely better off using binary transfer (in term of bandwidth) so I would personally stick to it.