pyvisa/pyvisa #829: [COM] Communication issue with Keysight DSOX-4024A using USB Connection

j-ruhi, 2024-06-07:
Instrument details
------------------
* Model: DSOX-4024A
* Communication: USB
* Link to the documentation (if available): https://www.keysight.com/us/en/assets/9018-06976/programming-guides/9018-06976.pdf

Output of `pyvisa-info`
-----------------------

I'm running a python program in a computer with Windows that is connected to a Keysight oscilloscope (DSOX-4024A) with an USB. Basically my program applies the write(...) to configure the oscilloscope so that the signal is inside the display and then reads some values with query(...), and then repeats this process a lot of times with different signals. The problem is that, sometimes when i execute the query(...) i get a Timeout error, but this doesn't happen everytime, it happens occasionally without following any apparent pattern.

The error is the following:
[image]
[image]

As you can see, the error always happens when i do the oscilloscope.query(...).

I checked the pyvisa documentation and then realised the problem may be that i had to define the termination_character, as i couldn't find it in the programming guide of this oscilloscope what character i had to use i am using '\n' for the moment, like this:

[image]

MatthieuDartiailh, 2024-06-07:
If the issue is transient, it is not a termination character issue.
What can happen is that the instrument is taking longer to process the operation you requested. It can sometimes take some time to an oscillscope to switch of measurement type for example.

j-ruhi, 2024-06-10:
And how can I solve that?

j-ruhi, 2024-06-10:
> If the issue is transient, it is not a termination character issue. What can happen is that the instrument is taking longer to process the operation you requested. It can sometimes take some time to an oscillscope to switch of measurement type for example.

And how can I solve that?

MatthieuDartiailh, 2024-06-10:
You can try to add some sleep, or recover graciously whenever a timeout occurs.