pyvisa/pyvisa #832: pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed. (Keysight E3633A)

ikarslio, 2024-06-19:
Hi I'm trying to connect my low voltage to Gui during the FC7 set up. I got the following error, and can't fix it. My LV source is Keysight E3633A. I don't have any connection problem with my high voltage (Keithley 2410). I need your help. Thank you so much for your help in advance!

ins = rm.open_resource("ASRL/dev/ttyUSB0::INSTR")
ins.baud_rate = 19200
ins.query('*IDN?')
Traceback (most recent call last):
File "", line 1, in
File "/home/phep/.local/lib/python3.9/site-packages/pyvisa/resources/messagebased.py", line 648, in query
return self.read()
File "/home/phep/.local/lib/python3.9/site-packages/pyvisa/resources/messagebased.py", line 486, in read
message = self._read_raw().decode(enco)
File "/home/phep/.local/lib/python3.9/site-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
chunk, status = self.visalib.read(self.session, size)
File "/home/phep/.local/lib/python3.9/site-packages/pyvisa_py/highlevel.py", line 520, in read
return data, self.handle_return_value(session, status_code)
File "/home/phep/.local/lib/python3.9/site-packages/pyvisa/highlevel.py", line 252, in handle_return_value
raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MatthieuDartiailh, 2024-06-19:
Could you please provide the information requested in the template (output of `pyvisa-info` and link to your instrument manual) ? Most likely causes are improper configuration of the serial parameters or termination characters.

ikarslio, 2024-06-19:
pyvisa-info
Machine Details:
   Platform ID:    Linux-5.14.0-427.16.1.el9_4.x86_64-x86_64-with-glibc2.34
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.9.18
   Compiler:       GCC 11.4.1 20231218 (Red Hat 11.4.1-3)
   Architecture:   ('x86', 64)
   Build:          Jan 24 2024 00:00:00 (#main)

PyVISA Version: 1.14.2.dev52+gdf628ff

Backends:
   ivi:
      Version: 1.14.2.dev52+gdf628ff (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1.post56+g4a433a5). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1.post56+g4a433a5). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      VICP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

https://www.keysight.com/us/en/assets/9018-01164/user-manuals/9018-01164.pdf

ploeness, 2024-06-27:
I also encountered this issue, and here's how I resolved it:

gpo3303.write("*IDN?")

data = b''
while True:
    data += gpo3303.read_bytes(1)
    if data.endswith(b'\r'):
        break
print(data)

MatthieuDartiailh, 2024-06-27:
Setting `read_termination` to either `"\r"` or `"\r\n"` should fix the issue then.

ikarslio, 2024-06-27:
Thank you for your help. Actually, I changed the cable into RS232 for the low voltage which is Keysight E3633E. I have the low voltage connection, but this time I lost the high voltage connection which is Keithley 2410. I get time same error. I tried these 2 solutions today that you post, but they didn't work. Do you have any other suggestions by chance? Thank you