pyvisa/pyvisa #840: [COM] Communication issue with HP 8510C using Agilent 82357B (GPIB to USB)

Aerotusk, 2024-07-26:
Instrument details
------------------
* Model: HP 8510C
* Communication: GPIB
* Link to the documentation: [https://www.keysight.com/us/en/assets/9018-40641/user-manuals/9018-40641.pdf](https://www.keysight.com/us/en/assets/9018-40641/user-manuals/9018-40641.pdf)

Output of `pyvisa-info`
Machine Details:
   Platform ID:    Windows-10-10.0.22631-SP0
   Processor:      Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\odann\AppData\Local\Programs\Python\Python311\python.exe
   Version:        3.11.2
   Compiler:       MSC v.1934 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Feb  7 2023 16:38:35 (#tags/v3.11.2:878ead1)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 25167104
         Spec. Version: 7340544
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 25167104
         Spec. Version: 7340544
-----------------------

Hello all!
I am trying to communicate with my (old) HP 8510C. I have both Keysight IO libraries and NI-Visa installed, and can see (and communicate with!) the unit at GPIB address 16 in both, but ResourceManager comes up empty:

[image]
[image]

My code:
import pyvisa
pyvisa.log_to_screen()
rm=pyvisa.ResourceManager("C:\\Windows\\System32\\visa64.dll")
print(rm.list_resources())

2024-07-25 23:11:20,590 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2024-07-25 23:11:20,596 - pyvisa - DEBUG - Created library wrapper for C:\Windows\System32\visa64.dll
2024-07-25 23:11:20,668 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x0000024E67530ED0>',) -> 0
2024-07-25 23:11:20,672 - pyvisa - DEBUG - Created ResourceManager with session 4097
2024-07-25 23:11:20,793 - pyvisa - DEBUG - viFindRsrc(4097, '?*::INSTR', '<ViObject object at 0x0000024E67531150>', 'c_ulong(0)', <ctypes.c_char_Array_256 object at 0x0000024E675313D0>) -> -1073807343
()
2024-07-25 23:11:21,007 - pyvisa - DEBUG - Closing ResourceManager (session: 4097)
2024-07-25 23:11:21,016 - pyvisa - DEBUG - viClose(4097,) -> 0

I have tried ResourceManager with and without specifying a path, outcome is the same. Any help is appreciated!

MatthieuDartiailh, 2024-07-26:
Do you have a PCI GPIB board or USB-GPIB adapter ? In the later from which manufacturer is the adapter ?

Aerotusk, 2024-07-26:
I'm using a USB-GPIB adapter, an Agilent 82357B. I have drivers installed for it and it shows up correctly in device manager, and I can send commands and receive data through Keysight Connection Expert and the NI-VISA tool for sending commands over GPIB

MatthieuDartiailh, 2024-07-26:
Typically USB-GPIB adapters only work with the VISA implementation of the adapter manufacturer. I do not know what Keysight and Ni explorers do that make it work but for PyVISA you likely need to force VISA to use the Keysight library for GPIB.

Keysight Connection expert provides a conflict manager that can be used to select a VISA library on a per resource basis. I suggest you use it to force the use of the Keysight implementation for GPIB.

Aerotusk, 2024-07-27:
That fixed it! Had to go through these steps since Keysight Connection Expert wouldn't change the implementation:

[https://docs.keysight.com/kkbopen/fixing-corrupted-visa-conflict-manager-state-on-dual-vendor-visa-configurations-749649840.html](https://docs.keysight.com/kkbopen/fixing-corrupted-visa-conflict-manager-state-on-dual-vendor-visa-configurations-749649840.html)

closed 2024-07-27 by Aerotusk

MatthieuDartiailh, 2024-07-27:
Happy you got things to work !