pyvisa/pyvisa #844: ValueError: gpib_ctypes is installed but could not locate the gpib library.

SengerM, 2024-08-13:
I just installed pyvisa in my Fedora 40 PC. There is a Keithley 2700 connected through the National Instruments GPIB-USB-A adapter into the PC. When I try to open the communication with the instrument, I get this error:

```python
Python 3.12.4 (main, Jun  7 2024, 00:00:00) [GCC 14.1.1 20240607 (Red Hat 14.1.1-5)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyvisa
>>> rm = pyvisa.ResourceManager('@py')
/home/someone/.local/lib/python3.12/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
>>> rm.open_resource('GPIB::16::INSTR')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/someone/.local/lib/python3.12/site-packages/pyvisa/highlevel.py", line 3292, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/someone/.local/lib/python3.12/site-packages/pyvisa/resources/resource.py", line 281, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/someone/.local/lib/python3.12/site-packages/pyvisa/highlevel.py", line 3217, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/someone/.local/lib/python3.12/site-packages/pyvisa_py/highlevel.py", line 168, in open
    sess = cls(session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/someone/.local/lib/python3.12/site-packages/pyvisa_py/sessions.py", line 861, in __init__
    raise ValueError(self.session_issue)
ValueError: gpib_ctypes is installed but could not locate the gpib library.
Please manually load it using:
  gpib_ctypes.gpib.gpib._load_lib(filename)
before importing pyvisa.
```

**Output of `pyvisa-info`**
```
$ pyvisa-info 
/home/someone/.local/lib/python3.12/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details:
   Platform ID:    Linux-6.9.12-200.fc40.x86_64-x86_64-with-glibc2.39
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.12.4
   Compiler:       GCC 14.1.1 20240607 (Red Hat 14.1.1-5)
   Architecture:   ('x86', 64)
   Build:          Jun  7 2024 00:00:00 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.2
      ASRL INSTR: Available via PySerial (3.5)
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available 
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         gpib_ctypes is installed but could not locate the gpib library.
         Please manually load it using:
           gpib_ctypes.gpib.gpib._load_lib(filename)
         before importing pyvisa.
      GPIB INTFC:
         gpib_ctypes is installed but could not locate the gpib library.
         Please manually load it using:
           gpib_ctypes.gpib.gpib._load_lib(filename)
         before importing pyvisa.
```

MatthieuDartiailh, 2024-08-13:
On linux, you need to install linux-gpib. Gpib- ctypes can be used instead of the native linux-gpib bindings or can be used to access National instruments gpib library if you manage to install it.

SengerM, 2024-08-14:
I have installed `gpib-ctypes` using `pip install gpib-ctypes` because there was a previous error message suggesting that `gpib-ctypes` has more functionality than `linux-gpib`. Do I still have to install `linux-gpib` (or the NI library) for `gpib-ctypes` to work?

arr-ee, 2024-08-14:
Yes, `gpib-ctypes` is just a python interface — you need something for it to interface to.

Error message should be clearer about that.

SengerM, 2024-08-14:
@arr-ee thanks for clarifying that. For the record, this is the error message after uninstalling `gpib-ctypes`:

```python
In [1]: import pyvisa
In [2]: pyvisa.ResourceManager().open_resource('GPIB::16::INSTR')

ValueError: Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
No module named 'gpib'
```

which maybe could be improved.

closed 2024-08-14 by SengerM