pyvisa/pyvisa #847: Using USBInteruptEvent

hinxx, 2024-09-02:
I'm working with the USB INSTR resource class and I'm trying to catch VI_EVENT_SERVICE_REQ event.

As seen here:

https://github.com/pyvisa/pyvisa/blob/5b1fcfd8f42dfaef4ad37389c7847e34e1c945a8/pyvisa/events.py#L196

I would like to access status, size and data from such event.

In my code I do this:
```
	inst.enable_event(pyvisa.constants.VI_EVENT_SERVICE_REQ, pyvisa.constants.VI_QUEUE)
	waitrsp = inst.wait_on_event(pyvisa.constants.VI_EVENT_SERVICE_REQ, 5000)
	print('event response event:', waitrsp.event, 'reasponse ret:',waitrsp.ret)
```
I see this printed:
```
event response event: <pyvisa.events.Event object at 0x7f9e45277c10> reasponse ret: StatusCode.success
```

The event is not of `USBInteruptEvent` type and does not hold the attributes I am looking for: status, size and data.

What am I doing wrong?

MatthieuDartiailh, 2024-09-03:
You are enabling the event queue for service request events as a consequence this is the kind of event you get in the queue. To do what you want you need to do the following:
```python
	inst.enable_event(pyvisa.constants.EventType.usb_interrupt, pyvisa.constants.VI_QUEUE)
	waitrsp = inst.wait_on_event(pyvisa.constants.EventType.usb_interrupt, 5000)
	print('event response event:', waitrsp.event, 'reasponse ret:',waitrsp.ret)
``` 
However I am not how exactly sure how to generate a USB interrupt.

hinxx, 2024-09-04:
Thanks I'll give that a try!

> However I am not how exactly sure how to generate a USB interrupt.

I've read somewhere that it would be generated if the device sends vendor (not standard) message on the interrupt in endpoint. Will need to try it out..

hinxx, 2024-09-04:
After making my USB device return `bNotify1` value of 0x40 (instead of 0x81) signifying vendor interrupt IN message I could see the expected data in pyvisa using the code you suggested.

closed 2024-09-04 by hinxx

MatthieuDartiailh, 2024-09-04:
Happy you got it to work.