pyvisa/pyvisa #858: Fails to find connected devices

drmcnelson, 2024-09-30:
Attempting to communicate with Siglent SPD1000x

list_resources()  returns nothing.

open fails.

Following are 

(a) the output from lsusb -v

(b) a sample program, downloaded from here or siglent (i forget which) with some added diagnostics

(c)  the stack trace from the program

 - - -
 **lsusb output**

Bus 001 Device 009: ID f4ec:1410 Atten Electronics / Siglent Technologies SPD1000X
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0xf4ec Atten Electronics / Siglent Technologies
  idProduct          0x1410 SPD1000X
  bcdDevice            1.00
  iManufacturer           1 Siglent
  iProduct                2 SPD1000X
  iSerial                 3 SPD13DCC8R0195
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       254 Application Specific Interface
      bInterfaceSubClass      3 Test and Measurement
      bInterfaceProtocol      1 TMC
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)

**sample program**

#
#JAC: 03/31/2020
#-Added \n to write and read terminations
 
import pyvisa
import time # for sleep
import binascii
 
def main():
    rm = pyvisa.ResourceManager()
    print( 'list resources' )
    rm.list_resources()
    print( 'instadd')
    instadd = 'USB0::0xF4EC::0x1410::SPD13DCC8R0195::INSTR'
    print( 'open')
    inst = rm.open_resource(instadd)
    inst.write_termination='\n'
    inst.read_termination='\n'
    print (rm.list_resources())
    time.sleep(0.04)
    inst.write('OUTP CH1,ON')
    time.sleep(2)
    inst.write('OUTP CH1,OFF')
    time.sleep(2)
    inst.write('*IDN?')
    #print ("here")
    time.sleep(1)
    qStr = inst.read()
    print (str(qStr))
    inst.close()
    
if __name__ == '__main__':
    main()

**output and stack trace**
list resources
instadd
open
Traceback (most recent call last):
  File "/home/datacollection/TeensyDataAcquistion/Python_Programs_240927/./SPD1000x.py", line 35, in <module>
    main()
  File "/home/datacollection/TeensyDataAcquistion/Python_Programs_240927/./SPD1000x.py", line 18, in main
    inst = rm.open_resource(instadd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa/highlevel.py", line 3292, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python3.12/site-packages/pyvisa/resources/resource.py", line 281, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa/highlevel.py", line 3217, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/highlevel.py", line 168, in open
    sess = cls(session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/sessions.py", line 322, in __init__
    self.after_parsing()
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/usb.py", line 82, in after_parsing
    self.interface = self._intf_cls(
                     ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/protocols/usbtmc.py", line 307, in __init__
    super(USBTMC, self).__init__(vendor, product, serial_number, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/protocols/usbtmc.py", line 202, in __init__
    raise ValueError("No device found.")
ValueError: No device found.

drmcnelson, 2024-09-30:
P/S Here is the output from pyvisa-info

Machine Details:
   Platform ID:    Linux-6.8.8-300.fc40.x86_64-x86_64-with-glibc2.39
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.12.3
   Compiler:       GCC 14.0.1 20240411 (Red Hat 14.0.1-0)
   Architecture:   ('x86', 64)
   Build:          Apr 17 2024 00:00:00 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      TCPIP SOCKET: Available 
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

drmcnelson, 2024-09-30:
Thank you, here is the output

drmcnelson, 2024-09-30:
Permissions is part of it,  here it is again with sudo

sudo ./SPD1000x.py
Resource Manager of Visa Library at py
list resources
('ASRL/dev/ttyS0::INSTR', 'ASRL/dev/ttyS1::INSTR', 'ASRL/dev/ttyS2::INSTR', 'ASRL/dev/ttyS3::INSTR', 'ASRL/dev/ttyS4::INSTR', 'ASRL/dev/ttyS5::INSTR', 'ASRL/dev/ttyS6::INSTR', 'ASRL/dev/ttyS7::INSTR', 'ASRL/dev/ttyS8::INSTR', 'ASRL/dev/ttyS9::INSTR', 'ASRL/dev/ttyS10::INSTR', 'ASRL/dev/ttyS11::INSTR', 'ASRL/dev/ttyS12::INSTR', 'ASRL/dev/ttyS13::INSTR', 'ASRL/dev/ttyS14::INSTR', 'ASRL/dev/ttyS15::INSTR', 'ASRL/dev/ttyS16::INSTR', 'ASRL/dev/ttyS17::INSTR', 'ASRL/dev/ttyS18::INSTR', 'ASRL/dev/ttyS19::INSTR', 'ASRL/dev/ttyS20::INSTR', 'ASRL/dev/ttyS21::INSTR', 'ASRL/dev/ttyS22::INSTR', 'ASRL/dev/ttyS23::INSTR', 'ASRL/dev/ttyS24::INSTR', 'ASRL/dev/ttyS25::INSTR', 'ASRL/dev/ttyS26::INSTR', 'ASRL/dev/ttyS27::INSTR', 'ASRL/dev/ttyS28::INSTR', 'ASRL/dev/ttyS29::INSTR', 'ASRL/dev/ttyS30::INSTR', 'ASRL/dev/ttyS31::INSTR', 'USB0::62700::5136::SPD13DCC8R0195\x00\x00\x00\x00::0::INSTR')
instadd
open
Traceback (most recent call last):
  File "/home/datacollection/TeensyDataAcquistion/Python_Programs_240927/./SPD1000x.py", line 36, in <module>
    main()
  File "/home/datacollection/TeensyDataAcquistion/Python_Programs_240927/./SPD1000x.py", line 19, in main
    inst = rm.open_resource(instadd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa/highlevel.py", line 3292, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python3.12/site-packages/pyvisa/resources/resource.py", line 281, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa/highlevel.py", line 3217, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/highlevel.py", line 168, in open
    sess = cls(session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/sessions.py", line 322, in __init__
    self.after_parsing()
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/usb.py", line 82, in after_parsing
    self.interface = self._intf_cls(
                     ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/protocols/usbtmc.py", line 307, in __init__
    super(USBTMC, self).__init__(vendor, product, serial_number, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/protocols/usbtmc.py", line 202, in __init__
    raise ValueError("No device found.")
ValueError: No device found.

arr-ee, 2024-09-30:
Have a look at this comment: https://github.com/pyvisa/pyvisa/issues/758#issuecomment-2041328096

I’ll try to pick up https://github.com/pyvisa/pyvisa-py/pull/423 soon unless somebody else does it, there’s minimal changes + docs to be done.

arr-ee, 2024-09-30:
Although I just noticed the null bytes in the address generated from root — that’s a bit odd.

drmcnelson, 2024-09-30:
The first comment,  got use past the permissions for the listing, here is where we are now, without sudo.

Resource Manager of Visa Library at py
list resources
('ASRL/dev/ttyS0::INSTR', 'ASRL/dev/ttyS1::INSTR', 'ASRL/dev/ttyS2::INSTR', 'ASRL/dev/ttyS3::INSTR', 'ASRL/dev/ttyS4::INSTR', 'ASRL/dev/ttyS5::INSTR', 'ASRL/dev/ttyS6::INSTR', 'ASRL/dev/ttyS7::INSTR', 'ASRL/dev/ttyS8::INSTR', 'ASRL/dev/ttyS9::INSTR', 'ASRL/dev/ttyS10::INSTR', 'ASRL/dev/ttyS11::INSTR', 'ASRL/dev/ttyS12::INSTR', 'ASRL/dev/ttyS13::INSTR', 'ASRL/dev/ttyS14::INSTR', 'ASRL/dev/ttyS15::INSTR', 'ASRL/dev/ttyS16::INSTR', 'ASRL/dev/ttyS17::INSTR', 'ASRL/dev/ttyS18::INSTR', 'ASRL/dev/ttyS19::INSTR', 'ASRL/dev/ttyS20::INSTR', 'ASRL/dev/ttyS21::INSTR', 'ASRL/dev/ttyS22::INSTR', 'ASRL/dev/ttyS23::INSTR', 'ASRL/dev/ttyS24::INSTR', 'ASRL/dev/ttyS25::INSTR', 'ASRL/dev/ttyS26::INSTR', 'ASRL/dev/ttyS27::INSTR', 'ASRL/dev/ttyS28::INSTR', 'ASRL/dev/ttyS29::INSTR', 'ASRL/dev/ttyS30::INSTR', 'ASRL/dev/ttyS31::INSTR', 'USB0::62700::5136::SPD13DCC8R0195\x00\x00\x00\x00::0::INSTR')
instadd
open
Traceback (most recent call last):
  File "/home/datacollection/TeensyDataAcquistion/Python_Programs_240927/./SPD1000x.py", line 36, in <module>
    main()
  File "/home/datacollection/TeensyDataAcquistion/Python_Programs_240927/./SPD1000x.py", line 19, in main
    inst = rm.open_resource(instadd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa/highlevel.py", line 3292, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python3.12/site-packages/pyvisa/resources/resource.py", line 281, in open
    self.session, status = self._resource_manager.open_bare_resource(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa/highlevel.py", line 3217, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/highlevel.py", line 168, in open
    sess = cls(session, resource_name, parsed, open_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/sessions.py", line 322, in __init__
    self.after_parsing()
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/usb.py", line 82, in after_parsing
    self.interface = self._intf_cls(
                     ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/protocols/usbtmc.py", line 307, in __init__
    super(USBTMC, self).__init__(vendor, product, serial_number, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/pyvisa_py/protocols/usbtmc.py", line 202, in __init__
    raise ValueError("No device found.")
ValueError: No device found.

drmcnelson, 2024-09-30:
Notice it is hung after the second list_resources.

So, take out the read termination, and voila, we get an answer.  It is incredibly slow, but it answers.

#!/usr/bin/python

#
#JAC: 03/31/2020
#-Added \n to write and read terminations
 
import pyvisa
import time # for sleep
import binascii
 
def main():
    rm = pyvisa.ResourceManager()
    print(rm)
    print( 'list resources' )
    print( rm.list_resources() )
    print( 'instadd')
    instadd = 'USB0::0xF4EC::0x1410::SPD13DCC8R0195\0\0\0\0::0::INSTR'
    print( 'open', instadd)
    inst = rm.open_resource(instadd)
    inst.write_termination='\n'
    #inst.read_termination='\n'
    print (rm.list_resources())
    time.sleep(0.04)
    inst.write('OUTP CH1,ON')
    time.sleep(2)
    inst.write('OUTP CH1,OFF')
    time.sleep(2)
    inst.write('*IDN?')
    #print ("here")
    time.sleep(1)
    qStr = inst.read()
    print (str(qStr))
    inst.close()
    
if __name__ == '__main__':
    main()

Resource Manager of Visa Library at py
list resources
('ASRL/dev/ttyS0::INSTR', 'ASRL/dev/ttyS1::INSTR', 'ASRL/dev/ttyS2::INSTR', 'ASRL/dev/ttyS3::INSTR', 'ASRL/dev/ttyS4::INSTR', 'ASRL/dev/ttyS5::INSTR', 'ASRL/dev/ttyS6::INSTR', 'ASRL/dev/ttyS7::INSTR', 'ASRL/dev/ttyS8::INSTR', 'ASRL/dev/ttyS9::INSTR', 'ASRL/dev/ttyS10::INSTR', 'ASRL/dev/ttyS11::INSTR', 'ASRL/dev/ttyS12::INSTR', 'ASRL/dev/ttyS13::INSTR', 'ASRL/dev/ttyS14::INSTR', 'ASRL/dev/ttyS15::INSTR', 'ASRL/dev/ttyS16::INSTR', 'ASRL/dev/ttyS17::INSTR', 'ASRL/dev/ttyS18::INSTR', 'ASRL/dev/ttyS19::INSTR', 'ASRL/dev/ttyS20::INSTR', 'ASRL/dev/ttyS21::INSTR', 'ASRL/dev/ttyS22::INSTR', 'ASRL/dev/ttyS23::INSTR', 'ASRL/dev/ttyS24::INSTR', 'ASRL/dev/ttyS25::INSTR', 'ASRL/dev/ttyS26::INSTR', 'ASRL/dev/ttyS27::INSTR', 'ASRL/dev/ttyS28::INSTR', 'ASRL/dev/ttyS29::INSTR', 'ASRL/dev/ttyS30::INSTR', 'ASRL/dev/ttyS31::INSTR', 'USB0::62700::5136::SPD13DCC8R0195\x00\x00\x00\x00::0::INSTR')
instadd
open USB0::0xF4EC::0x1410::SPD13DCC8R0195::0::INSTR
('ASRL/dev/ttyS0::INSTR', 'ASRL/dev/ttyS1::INSTR', 'ASRL/dev/ttyS2::INSTR', 'ASRL/dev/ttyS3::INSTR', 'ASRL/dev/ttyS4::INSTR', 'ASRL/dev/ttyS5::INSTR', 'ASRL/dev/ttyS6::INSTR', 'ASRL/dev/ttyS7::INSTR', 'ASRL/dev/ttyS8::INSTR', 'ASRL/dev/ttyS9::INSTR', 'ASRL/dev/ttyS10::INSTR', 'ASRL/dev/ttyS11::INSTR', 'ASRL/dev/ttyS12::INSTR', 'ASRL/dev/ttyS13::INSTR', 'ASRL/dev/ttyS14::INSTR', 'ASRL/dev/ttyS15::INSTR', 'ASRL/dev/ttyS16::INSTR', 'ASRL/dev/ttyS17::INSTR', 'ASRL/dev/ttyS18::INSTR', 'ASRL/dev/ttyS19::INSTR', 'ASRL/dev/ttyS20::INSTR', 'ASRL/dev/ttyS21::INSTR', 'ASRL/dev/ttyS22::INSTR', 'ASRL/dev/ttyS23::INSTR', 'ASRL/dev/ttyS24::INSTR', 'ASRL/dev/ttyS25::INSTR', 'ASRL/dev/ttyS26::INSTR', 'ASRL/dev/ttyS27::INSTR', 'ASRL/dev/ttyS28::INSTR', 'ASRL/dev/ttyS29::INSTR', 'ASRL/dev/ttyS30::INSTR', 'ASRL/dev/ttyS31::INSTR', 'USB0::62700::5136::SPD13DCC8R0195\x00\x00\x00\x00::0::INSTR')
Siglent

drmcnelson, 2024-09-30:
Okay, that gets me going.   Thank you.

Maybe some more detail on installation would help.

drmcnelson, 2024-09-30:
Here is the recipe

sudo pip install pyusb
sudo pip install zeroconf
sudo pip install pyvisa-py
sudo pip install pyvisa

sudo cp usbtmcdevices.rules /etc/udev/rules.d/

sudo udevadm control --reload-rules && sudo udevadm trigger

And here is a rules with a few devices
## USB Test Measurement Class (USBTMC) instruments ##

KERNEL=="usbtmc/*", MODE="0660", GROUP="dialout"
KERNEL=="usbtmc[0-9]*", MODE="0660", GROUP="dialout"

# ThorLabs PM100D laser power meter
SUBSYSTEM=="usb", ATTR{idVendor}=="1313", ATTR{idProduct}=="8078", GROUP="dialout", MODE="0660"

# Agilent 34450A 5 1/2 Digit Multimeter
SUBSYSTEM=="usb", ATTR{idVendor}=="0957", ATTR{idProduct}=="b318", GROUP="dialout", MODE="0660"

# Siglent power supply
SUBSYSTEM=="usb", ATTR{idVendor}=="f4ec", ATTR{idProduct}=="1410", GROUP="dialout", MODE="0660"

MatthieuDartiailh, 2024-10-01:
@drmcnelson @arr-ee feel free to suggest improvement to pyvisa-py documentation regarding proper configuration of USBTMC devices. There is room for improvement but I have too little first hand experience there to make any substantial change.

arr-ee, 2024-10-01:
@MatthieuDartiailh yeah I’ll pick up the PR mentioned above and will update docs since it is a requirement for the PR to make sense.