pyvisa/pyvisa #860: [COM] Communication issue with OWON XDS3204AE Oscilloscope trying to read signal data (USB)

Choxolatine, 2024-10-04:
Hi, 
I'm new to pyvisa and trying to automate experiments in electronics. I could get it to work on a first device but I'm stuck with my oscilloscope. It is connected via USB (USBTMC) and detected correctly. I can open it as a resource in pyvisa, and change a bunch of settings (channels display, timescale, etc.) through SCPI commands, which I found in [this manual](https://files.owon.com.cn/software/Application/SDS1000_Oscilloscopes_SCPI_Protocol.pdf). This document is made for the SDS series and my oscilloscopde is an XDS series model, but I couldn't find a more appropriate guide.

The problem shows up when I try to get the voltage data from the oscilloscope, which I assume should be done with `Owon.query(':DATA:WAVE:DEPMem:All?')` (see page 35 of the manual). But this gives the following error:
`UnicodeDecodeError: 'ascii' codec can't decode byte 0x87 in position 4: ordinal not in range(128)`
So I tried going with a `Owon.write` instruction first, and then a `read` which allows to change the encoding. I tried a few like `utf-8`, `utf-16`, `latin1` etc but I always get a similar error, or a series of bytes resulting from wrong decoding, e.g. with `utf-8`:
`UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc4 in position 0: invalid continuation byte`

I also tried to use `query_binary_values` and `read_raw`. 

Here is the output of `Owon.read_raw()` after the `Owon.write(':DATA:WAVE:DEPMem:All?')` command: 
`b'\xc4\xfe\xc4\xfe\xc4\xfe\xc4\xfe\x8c\xfd\xc4\xfe\x8c\xfd\xc4\xfe\xc4\xfe\xc4\  [...] \x8c\xfd\xc4\xfe\xc4\xfe\xc4\xfe'`

And here is the error of `Owon.query_binary_values('CH1?')`:
`ValueError: Could not find hash sign (#) indicating the start of the block. The block begin by bytearray(b'')`

The thing is I don't really know what data form to expect from the instrument after the `Owon.write(':DATA:WAVE:DEPMem:All?')` command, and I don't know what encoding it uses. The only details in the manual I found are (p. 35) _"The data point is recorded as 12-bit, a point uses two bytes, little-endian byte order"_ but it doesn't help me (the documentation states that little-endian is default in pyvisa so the endianness shouldn't be involved).

Does anyone know how I could solve this please ?

Instrument details
------------------
* Model: OWON XDS3204AE
* Communication: USBTMC
* Link to the documentation: [XDS3000_Four-Channel_Series_Oscilloscopes_USER_MANUAL](https://files.owon.com.cn//probook/XDS3000_Four-Channel_Series_Oscilloscopes_USER_MANUAL.pdf)
* Owon [SDS1000_Oscilloscopes_SCPI_Protocol](https://files.owon.com.cn/software/Application/SDS1000_Oscilloscopes_SCPI_Protocol.pdf)

Output of `pyvisa-info`
-----------------------
(base) C:\Users\adact>pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.22631-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 12, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\adact\anaconda3\python.exe
   Version:        3.11.5
   Compiler:       MSC v.1916 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Sep 11 2023 13:26:23 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 25167104
         Spec. Version: 7340544
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 25167104
         Spec. Version: 7340544

My code
-----------------------

rm = visa.ResourceManager()
Owon = rm.open_resource('USB0::0x5345::0x1235::2332018::INSTR')
Owon.write_termination = '\n'
Owon.read_termination = ''
Owon.query('*IDN?')
Owon.write(':ACQuire:MODE SAMPle')
Owon.write(':ACQuire:AVERage:NUM 16') 
Owon.write(':ACQuire:DEPMEM 100K') 
Owon.write(':HORIzontal:SCALe 200ns')
Owon.write(':CH1:DISPlay ON')
Owon.write(':CH2:DISPlay ON')
Owon.write(':DATA:WAVE:DEPMem:All?')
Owon.read(encoding='utf-8')
rm.close()

arr-ee, 2024-10-04:
https://files.owon.com.cn/software/Application/XDS3000_and_XDS2000_Dual-Channel_Series_Oscilloscopes_SCPI_Protocol.pdf appears to be the better fitting documentation for your scope.

Data you’re getting via `read_raw` looks right at the first glance. You need to parse the binary data yourself, likely using the data from `HEADer?` command. https://github.com/h2lab/scopy/blob/ce28961e706f60c2014a9207227365e3b9d1d326/scopy/scopes/owon_xds.py#L244 is an example of how to do that.

MatthieuDartiailh, 2024-10-04:
You should be able to use query_binary_values if you know how many points (N) to expect using:
- datatype = "H"
- is_big_endian=False
- header_fmt="empty"
- expect_termination=False
- data_points=N

arr-ee, 2024-10-04:
There’s a 4 byte (iirc) length indicator in the response, which should be handled also (discarded assuming it parses fine? Haven’t worked with binary data in VISA much)

Choxolatine, 2024-10-04:
Thanks a lot for your answers ! The solution of @MatthieuDartiailh seems to be working but not entirely reliably (meaning that some random data points are way off the curve, but maybe that comes from the scope?). I will look into the parsing method pointed out by @arr-ee, thanks ! 
I'm not sure what you mean by "follow `read_binary_values` logic" @MatthieuDartiailh, could you please expand a bit ? How is it different from using `query_binary_values` with the parameters you suggested ?

Choxolatine, 2024-10-10:
Hi, 
So I almost got it right by sending  `Owon.write(':DATA:WAVE:DEPMem:CH1?')` followed by `raw_data_CH1 = Owon.read_raw(200002)`. The conversion into signed short is handled as done in the code linked by @arr-ee., and the value is rescaled using the channel scale. 

However it seems that for some reason, the resource manager shuts down at some point, causing an error after a random number of measurements. I could reduce the frequency of this error by resetting (closing and re-opening) my `resource` scope object between each pair of reads, but it still happens sometimes (once every 100 operations I would say).

Finally, it also happens randomly that the retrieved data is erroneous, and i mean the raw data. The binary file makes no sense, and once converted this gives me outliers in the curve obtained with the series of measurements (see picture below). Curiously this only happens with the data from channel 2 of the scope, never channel 1, even though they are obtained with the same lines of code. Here is what it looks like:

Normal binary data (CH), 100 first elements: 
`b">\r\x03\x00\x18\xc7\x0c\xc4d\xd3\x18\xc9\x88\xd3P\xe1\x80\xe8\xcc\xf6T\x03T\r\x14 \x04`\xb8A\xc8'X?\xc8\x17\x9c\xe7\x9c\xd1\xe0\xe4\x04\xd5<\xc7D\xc4\xf0\xd2`\xc7X\xd4\x04\xe1\xa8\xe8\x18\xf9\x1c\x03\xbc\x0e\x0c#\xa8_\xd4@t*(@\x14\x18\xcc\xe8\xd4\xd34\xe4\xb8\xd6\xcc\xc8P\xc5\x9c\xd5\xcc\xc8\x94\xd64\xe2\\\xea\xec\xf9"`

Weird binary data (CH2), with a lot of identical bytes, which is very different from the scope signal, first 100 elements:
`b'>\r\x03\x00, ,\xf8\xc0\xf2\xa0$\x00Z\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xc4U\x00\x1c\\\xf4\xfc\xec|!\xc4[\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\xfc\x7f\x90V|!D\xf9\xb4\xf9<)\x10c\xfc\x7f\xfc\x7f\xfc\x7f'`

Is this a known problem ? Any idea why this would happen ?
Thanks a lot anyway, I would'nt have gotten so far without your help !

[image]

### Part of the code handling the measurement:
def TakeMeasurement(Oscilloscope, SAVE_BIN_DATA):
    Vertical_scales = ['1mv', '2mv', '5mv', '10mv', '20mv', '50mv', '100mv', '200mv', '500mv', '1v', '2v', '5v', '10v']
    Oscilloscope.write(':CH1:SCALe 20mv')
    Oscilloscope.write(':CH2:SCALe 20mv')
    Oscilloscope.write(':HORIzontal:SCALe 20.0ns')
    CH1_scale = 20 #mV/div
    CH2_scale = 20 #mV/div

    test_max_CH1 = Oscilloscope.query(':MEASUrement:CH1:MAX?')
    test_min_CH1 = Oscilloscope.query(':MEASUrement:CH1:MIN?')
    test_max_CH2 = Oscilloscope.query(':MEASUrement:CH2:MAX?')
    test_min_CH2 = Oscilloscope.query(':MEASUrement:CH2:MIN?')
    time.sleep(0.5)

    while '?' in test_max_CH1 or '?' in test_min_CH1 : #get the channel scale right to avoid saturation
        current_scale = Oscilloscope.query(':CH1:SCALe?')
        if '.' in current_scale:
            current_scale_code = current_scale.split('.')[0] + 'mv'
        else: current_scale_code = current_scale.split('mV')[0] + 'mv'
        new_scale = Vertical_scales[Vertical_scales.index(current_scale_code) + 1]
        CH1_scale = float(new_scale.split('mv')[0])
        Oscilloscope.write(':CH1:SCALe ' + new_scale)
        time.sleep(1)
        test_max_CH1 = Oscilloscope.query(':MEASUrement:CH1:MAX?')
        test_min_CH1 = Oscilloscope.query(':MEASUrement:CH1:MIN?')
        Oscilloscope.write(':TRIGger:SINGle:EDGE:SOURce CH1')
        Oscilloscope.write(':TRIGger:SINGle:EDGE:LEVel ' + str(float(test_max_CH1.split(': ')[1].split('mV')[0])*0.9) + 'mv')
        #time.sleep(1)
        
    Oscilloscope.write(':TRIGger:SINGle:EDGE:SOURce CH1')
    Oscilloscope.write(':TRIGger:SINGle:EDGE:LEVel ' + str(float(test_max_CH1.split(': ')[1].split('mV')[0])*0.9) + 'mv')
    

    while '?' in test_max_CH2 or '?' in test_min_CH2 : #get the channel scale right to avoid saturation
        current_scale = Oscilloscope.query(':CH2:SCALe?')
        if '.' in current_scale:
            current_scale_code = current_scale.split('.')[0] + 'mv'
        else: current_scale_code = current_scale.split('mV')[0] + 'mv'
        new_scale = Vertical_scales[Vertical_scales.index(current_scale_code) + 1]
        Oscilloscope.write(':CH2:SCALe ' + new_scale)
        CH2_scale = float(new_scale.split('mv')[0])
        time.sleep(1)
        test_max_CH2 = Oscilloscope.query(':MEASUrement:CH2:MAX?')
        test_min_CH2 = Oscilloscope.query(':MEASUrement:CH2:MIN?')
    
    time.sleep(.5)
    Oscilloscope.write(':DATA:WAVE:DEPMem:CH1?')
    raw_data_CH1 = Oscilloscope.read_raw(200002)
    time.sleep(.5)
    Oscilloscope.write(':DATA:WAVE:DEPMem:CH2?')
    raw_data_CH2 = Oscilloscope.read_raw(200002)
    time.sleep(.5)
    
    data_CH1 = np.array(rawdata_to_signed_shorts(raw_data_CH1))
    
    data_CH2 = np.array(rawdata_to_signed_shorts(raw_data_CH2)) 
    

    len1, len2 = len(data_CH1), len(data_CH2)
    if len1 != len2:
        L = np.min([len1, len2])
        data_CH1 = data_CH1[:L]
        data_CH2 = data_CH2[:L]
    if SAVE_BIN_DATA:
        return np.array([data_CH1/2**15*5*CH1_scale/1000, data_CH2/2**15*5*CH2_scale/1000]).transpose(), raw_data_CH1, raw_data_CH2 #convert values to V
    return np.array([data_CH1/2**15*5*CH1_scale/1000, data_CH2/2**15*5*CH2_scale/1000]).transpose()

arr-ee, 2024-10-10:
Glad to see you’ve made considerable progress!

> However it seems that for some reason, the resource manager shuts down at some point, causing an error after a random number of measurements. I could reduce the frequency of this error by resetting [...]

Configure `pyvisa` logger to dump at DEBUG level to a logfile somewhere and check if you get any errors/warnings around “shut down” (and feel free to post logs here as well)

> Finally, it also happens randomly that the retrieved data is erroneous, and i mean the raw data.

This smells like a OWON fw issue. They are not known for the quality of their software, and I’ve seen reports of other scopes of theirs having glitches in data. The fact that it happens only on the second channel reaffirm my suspicions here. There does not appear to be any publicly available firmware updates though.

Not sure if it’s fixable in any way besides running outlier detection and extrapolating a “fixed” value from surroundings/dropping the datapoint.

Choxolatine, 2024-10-11:
Ok, if that's the firmware i'll just go for dropping the datapoint probably. I also tried to use CH1 and CH3 or CH4 instead of CH1 and CH2, but... it's just worse for some reason, I systematically get a timeout error for a query on CH1. I really only replaced 'CH2' by 'CH3'/'CH4'  everywhere in my code. I'll put the logs below in case you're interested.

So I'm going to stick to CH1/CH2 for now, and drop the glitchy data. 
I realized that the resource manager shutdown error was caused by a misplaced line, where I would query to one of my instruments just after closing it, in an `except` block raised when the measurement series stops unexpectedly. So that's probably not the reason why the series sometime stops. Lately no series stopped mid-run though, so i'll check the logs again if this issue comes back.

Thank you so much !

### Code where the timeout happens
```
Oscilloscope.write(':CH1:SCALe 20mv')
Oscilloscope.write(':CH3:SCALe 20mv')
Oscilloscope.write(':HORIzontal:SCALe 20.0ns')
CH1_scale = 20 #mV/div
CH3_scale = 20 #mV/div

test_max_CH1 = Oscilloscope.query(':MEASUrement:CH1:MAX?')
test_min_CH1 = Oscilloscope.query(':MEASUrement:CH1:MIN?')
test_max_CH3 = Oscilloscope.query(':MEASUrement:CH3:MAX?')
test_min_CH3 = Oscilloscope.query(':MEASUrement:CH3:MIN?')
```

### Traceback
```
Traceback (most recent call last):
  File "c:\users\adact\documents\aithra\manips\automated_measurements_ch3.py", line 252, in TakesSeries
    Meas_data = TakeMeasurement(Owon, SAVE_BIN_DATA)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\users\adact\documents\aithra\manips\automated_measurements_ch3.py", line 133, in TakeMeasurement
    test_max_CH1 = Oscilloscope.query(':MEASUrement:CH1:MAX?')
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\adact\anaconda3\Lib\site-packages\pyvisa\resources\messagebased.py", line 647, in query
    return self.read()
           ^^^^^^^^^^^
  File "C:\Users\adact\anaconda3\Lib\site-packages\pyvisa\resources\messagebased.py", line 485, in read
    message = self._read_raw().decode(enco)
              ^^^^^^^^^^^^^^^^
  File "C:\Users\adact\anaconda3\Lib\site-packages\pyvisa\resources\messagebased.py", line 441, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\adact\anaconda3\Lib\site-packages\pyvisa\ctwrapper\functions.py", line 2336, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\adact\anaconda3\Lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\adact\anaconda3\Lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

### Pyvisa logs
```
2024-10-11 10:53:26,707 - pyvisa - DEBUG - USB0::0x5345::0x1235::2332018::0::INSTR - opening ...
2024-10-11 10:53:26,709 - pyvisa - DEBUG - viOpen(4097, 'USB0::0x5345::0x1235::2332018::0::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x00000217F17DDFD0>') -> 0
2024-10-11 10:53:26,710 - pyvisa - DEBUG - USB0::0x5345::0x1235::2332018::0::INSTR - is open with session 2
2024-10-11 10:53:26,710 - pyvisa - DEBUG - viSetAttribute(2, <ResourceAttribute.termchar: 1073676312>, 10) -> 0
2024-10-11 10:53:26,710 - pyvisa - DEBUG - viSetAttribute(2, <ResourceAttribute.termchar_enabled: 1073676344>, 0) -> 0
2024-10-11 10:53:26,712 - pyvisa - DEBUG - viWrite(2, b':ACQuire:MODE SAMPle\n', 21, 'c_ulong(21)') -> 0
2024-10-11 10:53:26,713 - pyvisa - DEBUG - viWrite(2, b':ACQuire:AVERage:NUM 16\n', 24, 'c_ulong(24)') -> 0
2024-10-11 10:53:26,713 - pyvisa - DEBUG - viWrite(2, b':ACQuire:DEPMem 100K\n', 21, 'c_ulong(21)') -> 0
2024-10-11 10:53:26,714 - pyvisa - DEBUG - viWrite(2, b':HORIzontal:SCALe 20.0ns\n', 25, 'c_ulong(25)') -> 0
2024-10-11 10:53:26,714 - pyvisa - DEBUG - viWrite(2, b':CH1:DISPlay ON\n', 16, 'c_ulong(16)') -> 0
2024-10-11 10:53:26,715 - pyvisa - DEBUG - viWrite(2, b':CH3:DISPlay ON\n', 16, 'c_ulong(16)') -> 0
2024-10-11 10:53:27,266 - pyvisa - DEBUG - viWrite(2, b':CH1:SCALe 20mv\n', 16, 'c_ulong(16)') -> 0
2024-10-11 10:53:27,271 - pyvisa - DEBUG - viWrite(2, b':CH3:SCALe 20mv\n', 16, 'c_ulong(16)') -> 0
2024-10-11 10:53:27,274 - pyvisa - DEBUG - viWrite(2, b':HORIzontal:SCALe 20.0ns\n', 25, 'c_ulong(25)') -> 0
2024-10-11 10:53:27,282 - pyvisa - DEBUG - viWrite(2, b':MEASUrement:CH1:MAX?\n', 22, 'c_ulong(22)') -> 0
2024-10-11 10:53:27,285 - pyvisa - DEBUG - USB0::0x5345::0x1235::2332018::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2024-10-11 10:53:31,306 - pyvisa - DEBUG - viRead(2, <ctypes.c_char_Array_20480 object at 0x00000217F186C3D0>, 20480, 'c_ulong(0)') -> -1073807339
2024-10-11 10:53:31,311 - pyvisa - DEBUG - USB0::0x5345::0x1235::2332018::0::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
```

Choxolatine, 2024-10-11:
Here is the error that interrupts some of the measurement series. It happens during the read_raw method:

```
Traceback (most recent call last):
  File "c:\users\adact\documents\aithra\manips\automated_measurements.py", line 252, in TakesSeries
    Meas_data = TakeMeasurement(Owon, SAVE_BIN_DATA)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\users\adact\documents\aithra\manips\automated_measurements.py", line 176, in TakeMeasurement
    raw_data_CH2 = Oscilloscope.read_raw(200002)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\adact\anaconda3\Lib\site-packages\pyvisa\resources\messagebased.py", line 404, in read_raw
    return bytes(self._read_raw(size))
                 ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\adact\anaconda3\Lib\site-packages\pyvisa\resources\messagebased.py", line 441, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\adact\anaconda3\Lib\site-packages\pyvisa\ctwrapper\functions.py", line 2336, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\adact\anaconda3\Lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\adact\anaconda3\Lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_INP_PROT_VIOL (-1073807305): Device reported an input protocol error during transfer.
```

MatthieuDartiailh, 2024-10-11:
Happy to see you are making some nice progress. 

I did not look in the details of what you did. But I think you want `read_bytes` instead of `read_raw` (you want a specific number of bytes no matter if a termination is encountered or not).