pyvisa/pyvisa #864: [COM] Communication issue with  a Pi4 using PyVisa to Instruments

saclif, 2024-10-30:
I have been trying to communicate from a Pi4 to two instruments using PyVisa but i can never get it to see the devices.
below i list my two instruments and the Python code I ran. 

Instrument details
------------------
* Model: Chroma 61602 and ITECH IT8500G+ Series
* Communication: USB on both

Python Code on Pi4
--------------
(env-psu) eng@psupi:~ $ python
Python 3.11.2 (main, Aug 26 2024, 07:20:54) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyvisa
>>> rm = pyvisa.ResourceManager()
>>> print(rm.list_resources())
()

Note i never see any devices ever recognized.

Output of `pyvisa-info`
-----------------------
(env-psu) eng@psupi:~ $ pyvisa-info
Machine Details:
   Platform ID:    Linux-6.6.51+rpt-rpi-v8-aarch64-with-glibc2.36
   Processor:      

Python:
   Implementation: CPython
   Executable:     /home/eng/env-psu/bin/python
   Version:        3.11.2
   Compiler:       GCC 12.2.0
   Architecture:   ('arm', 64)
   Build:          Aug 26 2024 07:20:54 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.7.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
         Resource discovery:
         - VXI-11: ok
         - hislip: ok
      VICP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

ll-O-ll, 2025-01-06:
are you sure the instruments are connected on the same network? also `pip install gpib-ctypes`

saclif, 2025-01-06:
Ahmed,
I was able to fix my issue by correcting the USB setup at booting.

ScottC