pyvisa/pyvisa #868: Intermittent disconnection with Keysight 34972A LXI Data Acquisition Unit over USB after ~4 hours

willcvolpe, 2024-11-06:
I am experiencing intermittent disconnections when using a Keysight 34972A LXI Data Acquisition Unit over USB, controlled via Python and PyVISA. The instrument disconnects after being connected for approximately 4 hours, resulting in a VisaIOError: VI_ERROR_TMO when sending a *IDN? query. Notably, PyVISA is able to locate, identify, and open a connection to the instrument successfully, even after the disconnection occurs. The issue arises only when attempting read/write/query operations, which fail with a timeout error. This issue is temporarily resolved by physically unplugging and re-plugging the USB cable, which is not feasible for a remote and automated setup.

**Steps to Reproduce**:
- Connect the Keysight 34972A to a Windows 10 machine via USB.
- Use PyVISA to control the instrument and send regular queries (e.g., *IDN?) every few minutes.
- Observe that after approximately 4 hours, the query fails with a timeout error, even though the instrument remains visible to PyVISA.

**Expected Behavior**: The connection should remain stable indefinitely without requiring a manual reconnect of the USB cable.

**Actual Behavio**r: After approximately 4 hours, any read/write/query operation times out with VisaIOError: VI_ERROR_TMO. However, PyVISA can still list and create an instance of the instrument; it is only the communication (query/write) that fails.

**Attempted Solutions**:
- Increased timeout setting.
- Called instrument.clear() before queries.
- Closed and reopened the VISA resource connection.
- Verified the USB power management settings to prevent sleep (all settings appear correct).

Instrument details
------------------
* Model: Keysight 34972A LXI Data Acquisition Unit
* Communication: USB
* Link to the documentation (if available): https://www.keysight.com/us/en/product/34972A/lxi-data-acquisition-data-logger-switch-unit.html

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Windows-10-10.0.22631-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     <REDACTED>
   Version:        3.11.0
   Compiler:       MSC v.1933 64 bit (AMD64)
   Bits:           64bit
   Build:          Oct 24 2022 18:26:48 (#main)
   Unicode:        UCS4

PyVISA Version: 1.13.0

Backends:
   ivi:
      Version: 1.13.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 25167872
         Spec. Version: 7340544
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 25167872
         Spec. Version: 7340544
   py:
      Version: 0.7.0
      ASRL INSTR: Available via PySerial (3.5)
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

wcvo, 2024-11-07:
**Update: ** This does not seem to be a py-visa issue. I see the same issue using NI-VISA Interactive Control - I can see and connect to the instrument but not write/read any commands.

MatthieuDartiailh, 2024-11-07:
Then you may get more help from a Keysight forum. Feel free to share any progress you make it may help somebody else.

afischer-sweepme, 2024-11-08:
Hi, just found your issue when searching for solutions for a very similar issue in our forum:
https://forum.sweep-me.net/t/keithley-smu-2450-timeout-error/82

I added some possible steps to resolve the problem that you could try as well. Happy to receive any feedback if something worked for you.