pyvisa/pyvisa #871: HOWTO ?? Read resource, then open that resource, without manual intervention ??

dazz100, 2024-11-18:
Hi
This is not a bug or feature request.  If the Discussions tab was enabled, I would put it there.  Most likely I will be showing off my ignorance of Python, but here goes:

What I am trying to do is list the available resouces(s) then open the resource(s).

At present, I list the resource(s), then manually copy/paste into the `open_resource` method.
This is shown in the code snippet below.

```
print("# Show the list of Resources")
rm = pyvisa.ResourceManager()
print(rm.list_resources())

print("# Opening the SigGen")
SigGen = rm.open_resource('USB0::0x03EB::0x2065::GPIB_02_24238323232351C01171::INSTR')
```

The above code snippet works, but it requires manual intervention.

Where there are multiple resources, I want to open them all, then link the code to the correct instrument via the correct USB-GPIB adapter.   I have multiple usb-gpib adapters so the aim is to make the adapters transparent so I can connect any adapter to any instrument and use the gpib address and instrument ID to link the code, via a adapter, to the correct instrument.  

I am guessing there is a simple solution, but I can't find it.  Any advice gratefully received.

MatthieuDartiailh, 2024-11-26:
Hi,
You could iterate over the list returned by list_resource, open all resources and store them in a dictionary mapping the pseudo GPIB address to the resource.

dazz100, 2024-11-27:
Hi
Thanks.  I will look into that.  This is undoubtedly an issue with my poor (almost nil) python coding skill.

dazz100, 2025-01-06:
Hi
I have figured out some code that will open a single connected adapter without manual intervention.   This will work where a user has multiple adapters that can be randomly matched to different instruments.  This code snippet will make the adapter choice agnostic.

```
print("# Show the list of Resources")
rm = pyvisa.ResourceManager()
print(rm.list_resources())
rs = rm.list_resources()

print("# Opening the usb:gpib adapter")
# Resource = GPIB-usb adapter 0.  Different for each adapter.
instr = rm.open_resource(rs[0], send_end=False,)
```
I haven't tried it with multiple adapters, but I would expect it will only connect to the first one it finds.   It could easily be iterated through the list to find multiple adapters, but this would require additional code to identify the instrument connected to each adapter.   

I am sure the hardened Python coders will yawning already but given my near zero knowledge of python, I am impressed  :-)

closed 2025-01-06 by dazz100