pyvisa/pyvisa #880: [COM] Communication issue with RIGOLDM3058 using USB

LertxundiJon, 2024-12-17:
Instrument details
------------------
* Model: Rigol DM3058
* Communication: USB
* Link to the documentation (if available): 

I am trying to automate some measurements using a DM3058, and for this purpose, I am using the script attached.

To debug the issues I am facing, I connected another Rigol device (DM3021) in parallel, where I am running the "same" script (adjusting the necessary references), and it works correctly. I am sharing both scripts with you so you can see what I mean.
[Scripts.zip](https://github.com/user-attachments/files/18163232/Scripts.zip)

The issue with the DM3058 occurs when executing instrument.read(). Up to that point, the script works correctly (e.g., retrieving the device ID with the *IDN? command), but executing this specific command causes an error that abruptly terminates the script.

I have manually tested each command from a Python console. When I execute instrument.write("MEAS:CURR?"), I receive a return value of "12". However, when I try to execute instrument.read(), I get the following error (reported in Output of pyvisa-info):

Using NI's VISA interface (via MAX), I am also able to correctly retrieve the ID of the DM3058. However, unlike the DM3021, where I can successfully read the measured value (MEAS:VOLT?\n command and MEAS:CURR?\n command), the DM3058 does not behave as expected.

I have also verified that the DM3058 is set to Rigol mode among the three available options (RIGOL|AGILENT|FLUKE).
Does anyone know what might be causing this behavior?
Thank you and best regards.

Output of `pyvisa-info`
-----------------------
Traceback (most recent call last):
  File "<python-input-8>", line 1, in <module>
    current = instrument.read()
  File "C:\Users\43436\AppData\Local\Programs\Python\Python313\Lib\site-packages\pyvisa\resources\messagebased.py", line 485, in read
    message = self._read_raw().decode(enco)
              ~~~~~~~~~~~~~~^^
  File "C:\Users\43436\AppData\Local\Programs\Python\Python313\Lib\site-packages\pyvisa\resources\messagebased.py", line 441, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
                    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\43436\AppData\Local\Programs\Python\Python313\Lib\site-packages\pyvisa\ctwrapper\functions.py", line 2336, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\43436\AppData\Local\Programs\Python\Python313\Lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
           ~~~~~~~~~~~~~~~~~~~~~~~~^^^^
^^^^^^^^^^^^^^^^
  File "C:\Users\43436\AppData\Local\Programs\Python\Python313\Lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

arr-ee, 2024-12-17:
> Using NI's VISA interface (via MAX), I am also able to correctly retrieve the ID of the DM3058. However, unlike the DM3021, where I can successfully read the measured value (MEAS:VOLT?\n command and MEAS:CURR?\n command), the DM3058 does not behave as expected.

If your commands do not work in NI MAX, then the issue is with your commands.

According to https://int.rigol.com/Images/DM3058_ProgrammingGuide_EN_tcm7-3381.pdf#page57, it appears that 3058 requires you to specify whether you’re measuring AC or DC: `:MEAS:CURR:DC?`

Can you give this a try?

LertxundiJon, 2024-12-17:
It works! I am sure that I already tested this, but I don't know why it did not work... At first, I started using these commands (adding the DC), and when I couldn’t get it to work, I connected the DM3021 and tried to replicate what I was successfully doing with the other instrument.
I’m not sure what was happening back then (I must have been doing something else wrong), but anyway, it’s working correctly now. Thank you!!!!!!

closed 2024-12-17 by MatthieuDartiailh