pyvisa/pyvisa #882: Wrapping Pyvisa to Create Instrument-Specific Classes

StevenGoode, 2024-12-25:
Years ago, I worked at a different employer than I do now and we had a couple real programmers (unlike me who is just a hack and an advanced beginner at Python) who understood pyvisa pretty well. They created an environment for the rest of us where there was a base instrument class for all our GPIB/USB instruments with common commands as methods (e.g., connect, *IDN?, close, etc.) and they created some specialized classes for some of our common instruments (Agilent/Keysight power supplies, vector network analyzers, Fluke DMMs, etc.) with commands unique to those types of instruments and we could create more classes for more of our instruments. Those instrument classes inherited the base class and they were imported into our software so we just had to do things like this to turn on a power supply:

```
import baseClass
import agilent_e36313a #class of commands quasi-unique to this model
ps_gpib_address = 15
####
ps1 = agilent_e36313a(ps_gpib_address)
ps1.turn_on_all_supplies()
####
```
(and so on)

In addition to the easy commands we created for ourselves, we had an element of abstraction and we could to some degree substitute similar instruments and get them to work without having to start all over. In any case, That was back with Python 2.7 and Pyvisa earlier than 1.5 for sure (not sure exactly what version). Obviously things are different now but I'm trying to recreate something similar at my current employer using Python 3.12.7 and Pyvisa 1.14. 

The first step is trying to wrap an instrument class around pyvisa to create a class with methods defined for each instrument. Probably due to my lack of knowledge of pyvisa, VISA and related topics, I'm having trouble doing this. This may be a really bad idea but I assumed that once I create an instrument object by doing the following:

```
import pyvisa
ps1Addr = 'USB0::10893::4610::MY610567462::0::INSTR'
ps1 = pyvisa.ResourceManager().open_resource(ps1Addr)
```

then I should be able to inherit (or something related) ps1 into another class with methods particular to it: programming voltages and currents, turning on and off the supplies, etc. without wiping out the pyvisa methods already there. I have spent many hours trying to figure this out with a lot of errors. I tried subclassing. No joy:

```
import pyvisa
class ps_funks():

    def PS1allOff(self):
        self.write('OUTP OFF,(@1:3)')   
        
    def PS1allOn(self):
        self.write('OUTP ON,(@1:3)')   
```

then the main:

```
import pyvisa
import ps_funks
from ps_funks import *

ps1Addr = "USB0::10893::4610::MY610567462::0::INSTR"
ps = pyvisa.ResourceManager().open_resource('{0}'.format(ps1Addr))

ps1 = ps_funks() # subclassing

# if you believe Google AI, this should work

ps1.PS1allOn()  #creates: AttributeError: 'ps_funks' object has no attribute 'write' 
ps1.PS1allOff() #creates: AttributeError: 'ps_funks' object has no attribute 'write'
```

I tried a few more things like Mixins (sorry, no example code to show right now) but that didn't work either. 

The best thing I could come up with that actually works is using setattr() but that still is slightly clumsy. First I create a separate file with functions that I would want to use, here called "ps_funcs":

```
import pyvisa
class ps_funcs(object):

    def PS1allOff(self):
      #Turn OFF all three outputs of the DC power supply 
        self.write('OUTP OFF,(@1:3)')   
        
    def PS1allOn(self):
       #Turn ON all three outputs of the DC power supply 
        self.write('OUTP ON,(@1:3)')   
```
then here is the main:

```
import pyvisa
import ps_funcs
from ps_funcs import *

ps1Addr = "USB0::10893::4610::MY610567462::0::INSTR"
ps = pyvisa.ResourceManager().open_resource('{0}'.format(ps1Addr))

setattr(ps,"PS1allOff", ps_funcs.PS1allOff)
setattr(ps,"PS1allOn", ps_funcs.PS1allOn)

ps1 = ps
ps1.PS1allOn(ps1) # this works. 
ps1.PS1allOff(ps1) # this works too
```

If I try to do something like this in any other way, it does not work and I'm not able to use the methods in "ps_funcs".

So, bottom line, **I can't be the only person trying to create their own classes for instruments by wrapping something around the pyvisa ResourceManager base class. Is there a better way to do the same thing? How do you do this?**

asteppke, 2025-01-03:
This more a question about how to structure an instrument driver than anything pyvisa specific. 

In principle one approach given your example would be to inject the instr into the class (untested):

````python
import pyvisa

class PSFunks:
    def __init__(self, instr):
        self.instr = instr

    def allOff(self):
        self.instr.write('OUTP OFF,(@1:3)')   
        
    def allOn(self):
        self.instr.write('OUTP ON,(@1:3)')  

ps1_addr = 'USB0::10893::4610::MY610567462::0::INSTR'
instr = pyvisa.ResourceManager().open_resource(ps1_addr)

ps1 = PSFunks(instr)
ps1.allOff()

````

StevenGoode, 2025-01-07:
Thanks for your reply.

Agreed that this is mostly about creating an instrument driver. I posted it in Pyvisa simply because I was having trouble getting even the simplest adaptations of "textbook" methods to work correctly, thus I began to think that there might be something specific to Pyvisa. With the benefit of ten days away from work, I now have the clarity to see why my original approach was not going to work because there was a disconnect between ps and ps1 (my mistake):

```
import pyvisa
import ps_funks
from ps_funks import *

ps1Addr = "USB0::10893::4610::MY610567462::0::INSTR"
ps = pyvisa.ResourceManager().open_resource('{0}'.format(ps1Addr))
ps1 = ps_funks() # subclassing   **#OOPS. PS1 and PS are not the same.**

# if you believe Google AI, this should work

ps1.PS1allOn()  #creates: AttributeError: 'ps_funks' object has no attribute 'write' 
ps1.PS1allOff() #creates: AttributeError: 'ps_funks' object has no attribute 'write'
```
In any case, your solution, though untested, actually _does_ work. Yours does what I was trying to do, which was make the the Pyvisa methods like "write", "query", etc. available to my method(s). 

In my defense, I will say that my earlier attempts recognized the need to inherit the ResourceManager methods as you have done here. My major mistake was tweaking the code enough that it was no longer inheriting the existing Pyvisa ResourceManager methods in the init as you have done. As I kept making revision attempts to make this work, I managed to get farther away from that. 

For completeness and in case someone comes across this later, I'm going to reproduce my working implementation as two separate files:

PSFunky.py

```
import pyvisa

class PSFunky:
    def __init__(self, instr):
        self.instr = instr

    def allOff(self):
        self.instr.write('OUTP OFF,(@1:3)')   
        
    def allOn(self):
        self.instr.write('OUTP ON,(@1:3)') 

```

then the main: 

```
import pyvisa
from PSFunky import * #needed if your custom methods will be in a separate file

ps1_addr = 'USB0::10893::4610::MY610567462::0::INSTR'
instr = pyvisa.ResourceManager().open_resource(ps1_addr)

ps1 = PSFunky(instr)
ps1.allOff()
# etc.
```

Thanks again.

davis-andy, 2025-03-06:
To go off of what David mentioned, also not as well versed as python as him so not sure what the metaclass does but if you want something similar then you can do what I did:

First, I have a folder with some basic settings and other files which I call my globals
I have a settings file the holds other things but this is part of it
```py
# globals/settings.py
import pyvisa
RESOURCE_MANAGER = pyvisa.ResourceManager()
```

then I have a folder that holds all the files for each manufacturer I use, but also contains the base Instruments.py file
```py
# instruments/Instruments.py
from globals.settings import RESOURCE_MANAGER

class Instr:
    # Initialize with resource
    def __init__(self, resource) -> None:
        self.instr = RESOURCE_MANAGER.open_resource(resource)
        self.instr.write_termination = '\n'
        self.instr.read_termination = '\n'
        self.instr.timeout = 10000  # set timeout to 10 seconds

    # Close connection
    def close(self) -> None:
        self.instr.close()

    # Reset instrument
    def reset(self) -> None:
        self.instr.write('*RST')
        # self.instr.opc()

    # Clear instrument
    def clear(self) -> None:
        self.instr.write('*CLS')

    # OPC command
    def opc(self) -> None:
        self.instr.query('*OPC?')

    # Make instrument wait
    def wait(self) -> None:
        self.instr.write('*WAI')

    # Retrieve instrument ID
    def get_id(self) -> str:
        return self.instr.query('*IDN?')

# these are here because I want to have a write/read/query function without having to type self.instr all the time

    # Write to the instrument
    def write(self, command: str) -> None:
        self.instr.write(command)

    # Read the instrument
    def read(self) -> str:
        return self.instr.read()

    # Query the instrument
    def query(self, command: str) -> str:
        return self.instr.query(command)
```

Now just reference this for each instrument you want to create a new class for
```py
# instruments/Fluke.py
from instruments.Instruments import Instr

class Fluke5700A(Instr):
    # Set unit to Operate
    def operate(self) -> None:
        self.instr.write('OPER')

    # Set unit to Standby
    def standby(self) -> None:
        self.instr.write('STBY')

    # Set Boost On
    def set_boost_on(self) -> None:
        self.instr.write('BOOST ON')

    # Set output DC volt
    def set_volt_dc_value(self, volt: float) -> None:
        self.instr.write(f'OUT {volt} V')
```

Then you have in your main file similar to David's
```py
from instruments.Fluke import Fluke5700A

f = Fluke5700A('GPIB0::16::INSTR')  # or whatever the resource may be
print(f.get_id())
f.set_volt_dc_value(10)
f.operate()
f.close()
```

sfentonn, 2025-05-09:
Along these lines, I was wondering does anyone know what happens if you create multiple ResourceManager instances? I'm trying to create a "connection manager" system for different connection types aside from visa and I want to make sure I'm not putting in more work than I need to keep one ResourceManager alive without using singletons lol