pyvisa/pyvisa #883: Writing binary bites

dazz100, 2025-01-06:
Hi
I am trying to write a gpib command that sets the frequency on a 1990's vintage emc receiver.
The command sets the DAC outputs that tune the VCOs.  Frequency control consists of two bytes, coarse and fine frequency control.  
The command is made up of 3 binary bytes in the following sequence (all values represented as decimal):
255
course_freq
fine_freq

where the two variables are bytes.  An example of a valid command sequence is:  255 100 50   (no separator character)

So my issue is that I can't get the pyvisa commands to work.  Undoubtably operator error because I am a python noob.  I have tried a lot of different things.

Based on my understanding of the documentation here:  [https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html](url)
in the section 'Writing Binary Values'  this should work:
```
# valid input command decimal values 
freq_cse = 150
freq_fine= 50
values = list[255, freq_cse, freq_fine]
instr.write_binary_values(values, datatype='d')
```
but it doesn't.  I get an error message:
`TypeError: MessageBaseResource.write_binary_values() missing 1 required positional argument: 'values'`

I have a "values" argument so I don't know why I am getting an error message and how to fix it.

MatthieuDartiailh, 2025-01-06:
Hi,
write_binary_values expect a custom text before the data (hence the error you see) and will by default add a header to the packed data which is not what you want. In your case, it will be simpler to just use: `instr.write_raw(struct.pack("BBB", 255, freq_cse, freq_fine)`

dazz100, 2025-01-08:
Hi
Just closing the loop on this issue. 

I tried the suggested write_raw command and the emc30 receiver did respond but would only set the lowest frequency in the range.  The frequency is tunable with the manual front panel control. 

The fact that the emc30 responded at all indicates it recognised the command.
Byte values can't go out of range, so that could not be an issue.
The working front panel control uses a different circuit to the not-working remote frequency control.
There is every indication the command is correct, but the emc30 response is not correct.

I am running with the hypothesis that there is a hardware fault in the emc30 remote frequency control circuit.
I have started going through the hardware diagnostics process.

I designed and made a GPIB breakout board that plugs directly into a HP Logic Analyser  or/and my HP 54645D Mixed signal oscilloscope that includes a basic logic analyser function.  This connects between the emc30 and my DIY usb-gpib adapter, an enhanced version of a design by Xyphro. I am using this to physically confirm that the emc30 is getting the right command.  

The GPIB breakout board includes a standard HP logic analyser compatible box header and a pin header to monitor and probe signals.
[image]

The breakout board is inserted between the usb-gpib adapter and the emc30 GPIB port.
[image]

The emc30 was purchased with optional gpib software, that ran on DOS and included a dongle to plug into the Centronics printer port.    I don't have the software or the dongle, but the manual gives me a lot of confidence that the emc30 firmware works well with gpib.  The software has a lot of functionality and this would require good firmware.  
[image]

closed 2025-01-08 by dazz100