pyvisa/pyvisa #885: Error using write_binary_values() to write Gaussian pulse

Xander-Ault, 2025-01-07:
Hi,
I am trying to write a Gaussian pulse generated in Python to a keysight 33600A waveform generator. The pulse appears fine in Python,
[image]
but when I attempt to use `write_binary_values(message = 'SOUR2:DATA:ARB:DAC RABI, ', values = pulse, datatype='e')` the output no longer resembles the Gaussian pulse. 
[image]
I have played around with the values for `datatype` and `is_big_endian` and while the output may change slightly, none of them resemble the original pulse we generate. I am, however, able to generate the correct pulse using `write('SOUR2:DATA:ARB RABI, ' + str)` where `str` is a comma-separated string containing the floating point values.  
[image]
we would like to be able to use the binary function instead due to the significant speed-up in writing binary values over ascii values. I have attempted to use `write_raw(msg.encode("ascii") + struct.pack("<4096d", *pulse))` as suggested elsewhere, but the AWG complains with an "invalid character" error (though there is no Python error).

DavidLutton, 2025-01-07:
I did something similar for loading a ARB on a Agilent 33120A
```py 
inst.write_binary_values(
    'DATA:DAC VOLATILE, ', pulse, 
    datatype='h', is_big_endian=True
)  # -2047 and +2047 on 33120A
```
Where 'h' is a [short](https://numpy.org/doc/stable/reference/arrays.scalars.html#numpy.short) which is integer -32,768 to +32,767
On your generator "The DAC codes go from -32,768 to +32,767 on both the 33500 Series and 33600 Series."
so 'h' should work, there may be a may be a way to use float (-1 to +1) but I didn't need to find it at the time

I've misplaced the complete demo for loading an particular ARB so I may have scaled the data then sent integers

Xander-Ault, 2025-01-07:
Scaling the floats and rounding to integer values then using `datatype='h'` did seem to fix the issue. Although, I am curious as to why it doesn't seem to work with the floating point datatypes

MatthieuDartiailh, 2025-01-21:
Usually AWG encode their waveform as DAC codes (hence integer) because they can for some of them change the output gain. Happy you got things to work.

closed 2025-01-21 by MatthieuDartiailh