pyvisa/pyvisa #891: [COM] Communication issue with RIGOL DG1022Z using PyVISA via USB

mshrey, 2025-02-17:
Instrument details
------------------
* Model: RIGOL DG1022Z
* Communication: via USB
* Link to the documentation (if available): https://www.batronix.com/pdf/Rigol/ProgrammingGuide/DG1022_ProgrammingGuide_EN.pdf 
* Refer to Chapter 3, example 3, to generate an user-defined arbitrary wave. 
* One caveat: the documentation is for RIGOL DG1022, while I am using a DG1022Z, but as far as I know, the later is just a slightly better version, and most of the things are the same.

The communication with the device is fine, everything works as expected, I can change the frequency, use the inbuilt functions, can change amplitude as well. Except, when I want to communicate a user defined arbitrary wave, it doesn't work. When I try to follow the exact instructions given in the [programming guide](https://www.batronix.com/pdf/Rigol/ProgrammingGuide/DG1022_ProgrammingGuide_EN.pdf) to generate such a wave, I am getting the following error:

> Error after DATA:DAC: -220,"Parameter error" 
> An error occurred: DATA:DAC command failed with error: -220,"Parameter error"

Here's a snippet of the python code that I am running 

```
import pyvisa

try:
    rm = pyvisa.ResourceManager()
    inst = rm.open_resource('USB0::0x1AB1::0x0642::DG1ZA244406036::INSTR')  # Replace with your address

    print(inst.query("*IDN?"))  # Check instrument ID

    # 1. Select user-defined arbitrary wave
    inst.write("FUNC USER")

    # 2. Set the frequency
    inst.write("FREQ 100000")

    # 3. Set the unit of amplitude
    inst.write("VOLT:UNIT VPP")

    # 4. Set the high level
    inst.write("VOLT:HIGH 2")

    # 5. Set the low level
    inst.write("VOLT:LOW -2")

    # 6. Load the waveform data (example from manual)
    data_command = "DATA:DAC VOLATILE,8192,16383,8192,0"
    print(f"Sending command: {data_command}")  # Print for debugging
    inst.write(data_command)

    # Check for errors after DATA:DAC
    error_code = inst.query("SYST:ERR?")
    if error_code != "+0":  # +0 means no error
        print(f"Error after DATA:DAC: {error_code}")
        raise Exception(f"DATA:DAC command failed with error: {error_code}")

    # 7. Output the waveform in volatile memory
    inst.write("FUNC:USER VOLATILE")  # Select the volatile waveform

    # 8. Enable the output
    inst.write("OUTP ON")

    print("User-defined waveform generated and output enabled.")

except pyvisa.VisaIOError as e:
    print(f"VISA I/O Error: {e}")
except Exception as e:
    print(f"An error occurred: {e}")
finally:
    if 'inst' in locals() and inst:
        inst.close()
    if 'rm' in locals() and rm:
        rm.close()
```

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Windows-10-10.0.26100-SP0
   Processor:      AMD64 Family 25 Model 68 Stepping 1, AuthenticAMD

Python:
   Implementation: CPython
   Executable:     C:\Users\Dirac\miniconda3\envs\pyvisa_env\python.exe
   Version:        3.9.21
   Compiler:       MSC v.1929 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Dec 11 2024 16:35:24 (#main)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136

Any insights into what is going on here would be appreciated. Let me know if you need any more information.

arr-ee, 2025-02-19:
Given that you’re not getting errors at protocol level and you’re not doing binary transfers, have you confirmed the command you’re sending is correct? Does it work via NI console, for example?

mshrey, 2025-02-25:
Hi, sorry for such a late reply, the command seems fine, it does work with another model from the same company, RIGOL DG4102.  And I don't have access to NI console, so I cannot test that.

arr-ee, 2025-02-25:
According to the manual, DG1000Z line appears to be quite different from DG1000 (or DG4000 for that matter): https://beyondmeasure.rigoltech.com/acton/attachment/1579/f-0493/1/-/-/-/-/DG1000Z%20Programming%20Guide.pdf

It seems you need at least 8 points in your waveform (despite docs still using the same values in notes to the command); I would also check if the new example from that document works.

Marmaduke1st, 2025-02-25:
I'm using a DG1062Z and a DG1022Z in my day job, no problem at at all with communication.

You need to tell what channel your wanting to use:
Syntax [:SOURce[<n>]]:VOLTage[:LEVel][:IMMediate]:HIGH
{<voltage>|MINimum|MAXimum}