pyvisa/pyvisa #904: [COM] Question: terminating read after timeout

fenugrec, 2025-04-19:
Hi,
this is not a pyvisa bug, but something I don't know how to handle cleanly.

On some instruments (TDS744A), some commands return an unknown length of data (and, doesn't use the ieee488 block format), e.g.
- screenshots ('HARDCOPY START')
- reading files from the internal floppy ('FILESYSTEM:PRINT')

These return binary data, so read termination characters cannot be used.
This instrument does assert EOI on the last byte, but that information is lost when using a Prologix adapter. And anyway, other instruments (e.g. non-SCPI protocols over rs232) there is sometimes nothing at all. I think it should be a criminal offence to not use any kind of blockdata header, but alas.
The 'correct' way to handle this then, is to end the transfer after an inter-byte timeout.

But, some issues : 
- pyvisa's read functions, when they timeout, discard the data
- unclear if pyvisa timeouts are for the entire transaction (I think it is), or 'timeout after last byte received'.  Just setting a huge 30-second timeout would be awful.
- sure, I can manually read_bytes(1) in a loop and control the timeout myself, but do I really need to reinvent this wheel ?

Example numbers: 
- creating a BMP screenshot : 300kB, with a few short <1-second pauses during transfer. Total ~ 25 seconds
- creating an EPS screenshot: 30kB, with a few long 2-4second pauses, total ~ 15 sec
- reading a file from floppy : arbitrary size, 4kB "chunks" with 2-4 sec pauses

I'm really hoping I just missed something in the docs, but it doesn't seem that way.
Any suggestions welcome. Thanks !

Instrument details
------------------
* Model: Tektronix TDS744A
* Communication: GPIB via usb-gpib prologix-compatible

MatthieuDartiailh, 2025-04-25:
You are right that currently PyVISA discard all data retrieved before a timeout (whose duration applies to the whole transaction). Technically we could expose the partial data in the exception I think but I have not looked in all corner cases.

If you know the size of the transaction in advance you can use read_bytes to retrieve a fixed number of bytes. This way you can get all the data in one go. If you only have an estimate you can combine this and the read_bytes(1) to speed up the process.

fenugrec, 2025-04-26:
Thanks Matthieu.

> we could expose the partial data in the exception

that would make things easier indeed. , but I understand how it can be non-trivial with such a wide variety of backends.

This wasn't a bug report to begin with, will close now.

closed 2025-04-26 by fenugrec