pyvisa/pyvisa #911: Support for extended block lengths

spevs, 2025-05-19:
The IEEE 488.2 definite-length block format specifies the length of the block (number of bytes) with a decimal number in ASCII. This is preceded by a single decimal digit (0-9) which indicates how many digits long the length is. This means blocks cannot be 1 GB or larger, since the length indicator is limited to 9 decimal digits and hence the maximum representable value is 999,999,999 bytes.

As high end equipment has now reached these levels of memory, manufacturers have extended this system in various ways.

LeCroy and Tektronix have extended the length digit into hexadecimal, so the maximum length is now 15 digits (allowing up to 1 PB). Rohde & Schwarz have taken a different approach; they remove the initial length digit, and instead delimit the block length with parentheses, when the block length reaches 1 GB or larger. So for a 1,234,567,890 byte block (10-digit length) in these two systems, the header would be:

- `#A1234567890DATA` (Tektronix / LeCroy)
- `#(1234567890)DATA` (Rohde & Schwarz)

(At the moment it seems Keysight does not support ≥1 GB block lengths at all yet, according to the latest version manuals on their website, but I suspect they would follow the same path as Tek and LeCroy if they ever add support for it).

The Tek/LeCroy hexadecimal extension is supported easily enough by simply changing `int(header_length)` to `int(header_length, base = 16)` when evaluating the length digit, all previously valid inputs (using only decimal digits 0-9) will have the same value when interpreted in hex.

The R&S system requires some more work. I have written an implementation available in a fork here, modifying `parse_ieee_block_header`: https://github.com/spevs/pyvisa/blob/extended-block-length/pyvisa/util.py

I have tested this with a R&S scope with 1 GSa of memory and it works. I also tested that normal-sized block reads from various other instruments (not using any extended length systems) still work as expected. Unfortunately I do not have access to any Tek or LeCroy scopes with large enough sample memory that the extended length hexadecimal digit would come into play. But I am quite sure the `base = 16` would do the trick, that one seems hard to go wrong.

So far I have not covered block writes, which work the same way for the respective manufacturers. Right now this would require some user interface change as the user would need some way to specify which extended length syntax to use, unless it were to be auto-detected based on manufacturer in the IDN (which feels quite hacky). Ideally R&S should add support for accepting writes in the hexadecimal length format in addition to their parentheses format, which I will ask them to do. Most likely they would keep reads the same in order not to break existing code, but if they can at least accept writes in either format, it would eliminate the need for the user to have to specify the format, as the hex extension could be used for all writes, and reads can be auto-detected.

spevs, 2025-05-20:
> Since there is no support for writing yet, I will limit my response to reading.
> 
> Regarding the case of Lecroy/Tek, I think it would be mostly fine to attempt conversion to hexadecimal. The only thing that worries me is that it conflict with the old HP block header which starts with `#A`. Regarding the R&S case, we can add a new format (we already have `"ieee"`, `"hp"` and `"empty"`).
> 
> Can you make a PR ?

Sure I don't mind, but I don't think I have the ability to make new branches.

MatthieuDartiailh, 2025-05-20:
The usual Github workflow is to fork and then create a pull request to integrate changes from your fork in the original repo.

spevs, 2025-05-20:
> The usual Github workflow is to fork and then create a pull request to integrate changes from your fork in the original repo.

Thanks, I see it now. I had missed the "compare across forks" button. I'll modify the implementation to split it into a new 'rs' format and make a PR tomorrow.

closed 2025-06-10 by MatthieuDartiailh