pyvisa/pyvisa #916: [COM] Communication issue with Quicksyn FSW-0020 using PyVISA via USB

MilanQubic, 2025-05-29:
Instrument details
------------------
* Model: Phase Matrix Quicksyn FSW-0020
* Communication: USB
* Link to the documentation: [https://www.manualslib.com/manual/1485058/National-Instruments-Quicksyn-Fsw-0010-Quicksyn-Fsw-0020.html](https://www.manualslib.com/manual/1485058/National-Instruments-Quicksyn-Fsw-0010-Quicksyn-Fsw-0020.html)

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Windows-11-10.0.26100-SP0
   Processor:      Intel64 Family 6 Model 151 Stepping 2, GenuineIntel
Python:
   Implementation: CPython
   Executable:     C:\Users\Science\Documents\GitHub\TWPA\.venv\Scripts\python.exe
   Version:        3.13.0
   Compiler:       MSC v.1941 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Oct  7 2024 09:38:07 (#tags/v3.13.0:60403a5)
   Unicode:        UCS4
PyVISA Version: 1.15.0
Backends:
   ivi:
      Version: 1.15.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: Rohde & Schwarz GmbH
         Impl. Version: 5245960
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: Rohde & Schwarz GmbH
         Impl. Version: 5245960
         Spec. Version: 5244928
   py:
      Version: 0.8.0
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: ok
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available
      PRLGX_TCPIP INTFC: Available
      GPIB INSTR: Available
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      VICP INSTR:
         Please install PyVICP to use this resource type.
      PRLGX_ASRL INTFC:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

Hey everyone,

I've been trying to make PyVISA work with a Phase Matrix Quicksyn model FSW-0020 through USB. Whenever I try to read or query (I'm assuming that query is affected by extension of the problem with read), I receive a timeout error. Interestingly enough, I'm able to communicate with the Quicksyn perfectly fine through the NI-VISA Interactive Control program using SCPI commands. PyVISA also appears to be working perfectly fine with other instruments in our lab.

I've tried looking around about this problem already, and I saw some mentions of adding a '\n' to the query command, but I've had no luck. My code:

Hardware_VISA_Control.py
-----------------------
```
import pyvisa as visa
import time

class VISAHardware():
    def __init__(self, VISA_ADDRESS: str):
        """
        :param address: Address of the VNA device
        """
        self.VISA_ADDRESS = VISA_ADDRESS

        rm = visa.ResourceManager()
        #Instantiate HardWare object
        self.HW = rm.open_resource(self.VISA_ADDRESS)

        # Reset System
        self.HW.write('*RST')

    def getID(self):
        return self.HW.query('*IDN?')

class Quicksyn(VISAHardware):
    def __init__(self, VISA_ADRESS: str):
        super().__init__(VISA_ADRESS)

    def setFrequency(self, frequency: float):
        self.HW.write("FREQ %dGHz" %frequency) # Please enter a number in GHz

    def getFrequency(self):
        return self.HW.query("FREQ?")

    def setPower(self, power: float):
        self.HW.write("POW %d" %power) # Parameter is in dB

    def getPower(self):
        return self.HW.query("POW?")
```

VISA_List.py
-----------------------
```
import pyvisa as visa
import Hardware_VISA_Control

rm = visa.ResourceManager()
print(rm.list_resources())

VISA_ADDRESS = 'ASRL3::INSTR'
quicksyn_hopefully = Hardware_VISA_Control.Quicksyn(VISA_ADRESS=VISA_ADDRESS)
print(quicksyn_hopefully.getID())
```

Error for running VISA_List.py
-----------------------
C:\Users\Science\Documents\GitHub\TWPA\VISA_List.py
('ASRL3::INSTR', 'TCPIP0::169.254.175.56::hislip0::INSTR', 'USB0::0x0AAD::0x0023::123456::0::INSTR', 'USB0::0x0AAD::0x0088::111372::0::INSTR', 'USB0::0x0AAD::0x0088::113621::0::INSTR', 'USB0::0x0AAD::0x0088::113623::0::INSTR', 'USB0::0x2012::0x0017::0142::0::INSTR', 'TCPIP0::129.97.41.217::inst0::INSTR', 'TCPIP0::K-Instr0000.local::hislip0::INSTR', 'TCPIP0::K-Instr0000.local::inst0::INSTR')
Traceback (most recent call last):
  File "C:\Users\Science\Documents\GitHub\TWPA\VISA_List.py", line 9, in <module>
    print(quicksyn_hopefully.getID())
          ~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "C:\Users\Science\Documents\GitHub\TWPA\Hardware_VISA_Control.py", line 22, in getID
    return self.HW.query('*IDN?')
           ~~~~~~~~~~~~~^^^^^^^^^
  File "C:\Users\Science\Documents\GitHub\TWPA\.venv\Lib\site-packages\pyvisa\resources\messagebased.py", line 690, in query
    return self.read()
           ~~~~~~~~~^^
  File "C:\Users\Science\Documents\GitHub\TWPA\.venv\Lib\site-packages\pyvisa\resources\messagebased.py", line 519, in read
    message = self._read_raw().decode(enco)
              ~~~~~~~~~~~~~~^^
  File "C:\Users\Science\Documents\GitHub\TWPA\.venv\Lib\site-packages\pyvisa\resources\messagebased.py", line 473, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
                    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Science\Documents\GitHub\TWPA\.venv\Lib\site-packages\pyvisa\ctwrapper\functions.py", line 2339, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\Science\Documents\GitHub\TWPA\.venv\Lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 227, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
           ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Science\Documents\GitHub\TWPA\.venv\Lib\site-packages\pyvisa\highlevel.py", line 252, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Process finished with exit code 1

If anyone has any suggestions on what can be done to help with this, it would be much appreciated.

MatthieuDartiailh, 2025-05-30:
The hardware link is USB but the protocol is serial. Over such a link you need to get the baudrate and a couple extra parameters right inaddition to read and write termination.
Giving a quick look at the manual it does not describe remote communication beyond the use of the provided UI or I missed something.
Where did you find applicable commands for the instruments ?

MatthieuDartiailh, 2025-06-02:
Can you send a command that has a visible impact ? To determine if we have an issue just reading or also writing ?
Another option is to use read_bytes to retrieve the response one byte at a time to check if the instrument sent anything back.

MilanQubic, 2025-06-02:
We tried running an experiment using just write commands to adjust the Quicksyn's frequency and power output (FREQ 8, POW 0, OUTP:STATE ON). The output of the experiment was as if the commands never went through at all. It seems that while write commands don't return errors, they are essentially ignored.

Unfortunately, while switching between using the GUI and code control, it seems like NI-VISA Instrument Control has become unresponsive as well (though it still doesn't produce an error). We will running an experiment with NI-VISA Instrument Control tomorrow after a restart of the lab computer.

Trying the read_bytes command once again produces the same error.

closed 2025-06-04 by MilanQubic