pyvisa/pyvisa #920: Timeout functionality for read()

RiyaOfTheSpectra, 2025-06-18:
Is there a way to implement a timeout in the `instr.read()` function, to handle unresponsive instruments?

Currently, calling `instr.read()` will never return if the instrument is not responding or has no lines to write.

MatthieuDartiailh, 2025-06-19:
read does implement a time out mechanism. Can you share more details of your application ?

RiyaOfTheSpectra, 2025-06-19:
I couldn’t find anything about timeout in the documentation.

I’m communicating with a power supply ([https://www.scientificindia.com/product-details/1600-w-programmable-dc-power-supplies](url)), that sends multiline replies. Some lines end in a CR-LF, while some only end in an LF. (Abominable, I know.) And I can’t anticipate exactly how many lines will be there in the reply, so I want to loop `instr.read()` or `instr.read_raw()`, until I get a timeout longer than, say 0.1s, because for all it’s misgivings, the power supply does respond within a hundred milliseconds or so.

RiyaOfTheSpectra, 2025-06-19:
Oh thanks! I don’t know how I missed it. Let me try this and report back.

closed 2025-06-23 by RiyaOfTheSpectra

MatthieuDartiailh, 2025-06-23:
Happy you got things to work.